google/cameraview #233: Why TextureView is created as preview view when api level >= 24 ?

Sejen27, 2018-02-14:
`    private PreviewImpl createPreviewImpl(Context context) {
        PreviewImpl preview;
        if (Build.VERSION.SDK_INT < 14) {
            preview = new SurfaceViewPreview(context, this);
        } else {
            preview = new TextureViewPreview(context, this);
        }
        return preview;
    }`

API Level | Camera API | Preview View
-- | -- | --
9-13 | Camera1 | SurfaceView
14-20 | Camera1 | TextureView
21-23 | Camera2 | TextureView
24 | Camera2 | SurfaceView

DiFilosofa, 2018-07-12:
same problem here, crash on 8.0 android  because of using TexttureView
`TextureView doesn't support displaying a background drawable`

eviansplash, 2018-07-15:
any answer to this yet?

jimmysuncpt, 2018-12-17:
```
    private PreviewImpl createPreviewImpl(Context context) {
        PreviewImpl preview;
        if (Build.VERSION.SDK_INT < 14) {
            preview = new SurfaceViewPreview(context, this);
        } else {
            preview = new TextureViewPreview(context, this);
        }
        return preview;
    }
```
改成这样就好了：
```
    private PreviewImpl createPreviewImpl(Context context) {
        PreviewImpl preview;
        if (Build.VERSION.SDK_INT < 14) {
            preview = new SurfaceViewPreview(context, this);
        } else if (Build.VERSION.SDK_INT < 24){
            preview = new TextureViewPreview(context, this);
        } else {
            preview = new SurfaceViewPreview(context, this);
        }
        return preview;
    }
```