google/cameraview #234: camera view lag showing when call mCamera.stop()

ashish0121, 2018-02-23:
The cameraview object is causing lag when used in a view pager.
When scrolling back from cameraview to anothertab, the transition isnt smooth.
Can you please suggest if there is a way to handle it smoothly.

kelvinwatson, 2018-10-31:
@ashish0121 Any update on this?

ashish0121, 2018-10-31:
1) In view pager when ViewPager.SimpleOnPageChangeListener() function onPageSelected is called, then you can call in your mCamera.stop() if it doesnt match with camera preview page.
2) The view pager was still causing jerk but to a lesser extent. So it was kind of a partial fix.
3) You can try combination of coordinator layout with view pager though i haven't tried it myself.

kelvinwatson, 2018-10-31:
Yes that is what I'm doing, except I'm starting up another camera in another page, so I call mCamera1.stop() for Fragment1, and mCamera2.start() for Fragment2.

kelvinwatson, 2018-10-31:
I was able to resolve this using an async task:
```
private static class StopStartCameraAsyncTask extends AsyncTask<Void, Void, Void>
	{
		private final CameraView cameraToStop;
		private final CameraView cameraToStart;

		StopCameraTask(@NonNull CameraViewListener cameraToStop, @NonNull CameraViewListener cameraToStart)
		{
			this.cameraToStart = cameraToStart;
			this.cameraToStop = cameraToStop;
		}

		@Override
		protected Void doInBackground(Void... voids)
		{
			cameraToStop.stopCamera();
			return null;
		}

		@Override
		protected void onPostExecute(Void aVoid)
		{
			super.onPostExecute(aVoid);
			cameraToStart.startCamera();
		}
	}
```