google/cameraview #241: OutOfMemory onCameraOpened

OleksandrGrument, 2018-05-21:


cugxy, 2018-05-21:
？？？

OleksandrGrument, 2018-10-08:
Still have this bug, for now it occurs on Samsung Galaxy devices with 5 android

HemanthImandi, 2019-08-05:
In HUAWEI and samsung devices mostly this crash happening. Any help.

OleksandrGrument, 2019-08-05:
As since the cameraview is deprecated and no longer support, I suggest all migrate to camerax 

Here is some base examples including cameraview.
https://github.com/android/camera

closed 2019-08-05 by OleksandrGrument

HemanthImandi, 2019-08-06:
In camerax library  the minsdkversion is 21. But i have to use the camera in very low end devices(minsdkversion 16). Any other solution?

OleksandrGrument, 2019-08-06:
Yes, the other solution is to check your application on memory leaks. The OutOfMemory is a marker of a possible memory leak in application. 

https://riggaroo.co.za/fixing-memory-leaks-in-android-outofmemoryerror/

The easiest way to find leaks is to use LeakCanary.