google/cameraview #250: Implemented with Fragment but when minimize the App, camera preview showing black screen.

amit016verma, 2018-08-25:
Thanku for this demo , its help me alot , but one problem I am facing with that , I have implemented this with the fragment , its working fine everything but when I minimize my app and again open , not showing camera preview and when I click on capture its take the photo easily thats every strange can you help me out from this,

Thanks in Advance .

dbq4130, 2018-11-02:
CameraView：
//        if (Build.VERSION.SDK_INT < 21) {
            mImpl = new Camera1(mCallbacks, preview);
//        } else if (Build.VERSION.SDK_INT < 23) {
//            mImpl = new Camera2(mCallbacks, preview, context);
//        } else {
//            mImpl = new Camera2Api23(mCallbacks, preview, context);
//        }

Bhumik, 2019-02-14:
It happens in my device also. I am using motorola g5 plus.
When i switch task and again come back to preview screen, it starts showing green solid color screen.

Solution suggested by @dbq4130 is not useful as here we are discarding camera2 api at all.

dbq4130, 2019-02-18:
Thank you. I forgot to paste that code and found the reason.  it is createPreviewImpl()