google/cameraview #261: java.lang.NullPointerException@android.hardware.camera2.CameraCaptureSession.capture

jjn2004, 2018-11-27:
java.lang.NullPointerException: Attempt to invoke virtual method 'int android.hardware.camera2.CameraCaptureSession.capture(android.hardware.camera2.CaptureRequest, android.hardware.camera2.CameraCaptureSession$CaptureCallback, android.os.Handler)' on a null object reference
41
    at com.google.android.cameraview.Camera2.lockFocus(Camera2.java:29)
42
    at com.google.android.cameraview.Camera2.takePicture(Camera2.java:4)
43
    at com.google.android.cameraview.CameraView.takePicture(CameraView.java:2)
44
    at com.louisgeek.louiscustomcamera.CameraActivity.onClick(CameraActivity.java:14)
45
    at java.lang.reflect.Method.invoke(Method.java)

I run into the issue on HUAWEI 8.0.0 phones, EDI-AL10 and  HWI-AL00, 8.0.0     does anyone have  solution ?

pansaipan, 2018-12-17:
我也碰到这个问题

AvikSh14, 2019-01-29:
I faced this problem too in a Samsung device. In other devices camera works fine. What is the reason behind this problem and how to solve it?

ngocphung247, 2019-06-12:
Can you try add Thread.sleep(200) before call capture

chinahope, 2019-07-16:
I faced thie problem too in android9 vivo v1831A

kkhuluwa, 2019-10-18:
> I faced thie problem too in android9 vivo v1831A

I faced this too in vivo A9

prajwalmw, 2021-01-03:
Any solution anyone found on this guys?