google/cameraview #276: Caused by: java.lang.RuntimeException: Fail to connect to camera service

zhaoyangxihui, 2019-06-21:


zhaoyangxihui, 2019-06-21:
at android.hardware.Camera.<init>(Camera.java:575)
        at android.hardware.Camera.open(Camera.java:404)
        at com.google.android.cameraview.Camera1.openCamera(Camera1.java:276)
        at com.google.android.cameraview.Camera1.start(Camera1.java:89)
        at com.google.android.cameraview.CameraView.start(CameraView.java:250)
        at com.eva.gen3.awake.activity.set.panel.advanced.CustomCameraTestActivity.onResume(CustomCameraTestActivity.java:90)
        at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1270)
        at android.app.Activity.performResume(Activity.java:6827)

closed 2019-06-28 by zhaoyangxihui

ghiyatshanif, 2019-07-08:
how did you solve this issue?